Seeed-Studio/pi-hats #4: CAN hat cannot send messages

JuanAsturias, 2019-08-31:
Attempting to use cangen can0 -mv there are the following error messages reported by dmesg:

dmesg | grep can 
[    3.643575] mcp25xxfd_can: loading out-of-tree module taints kernel.
[   51.220770] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   94.347674] can: controller area network core (rev 20170425 abi 9)
[   94.367269] can: raw protocol (rev 20170425)
[   94.370485] mcp25xxfd spi0.0 can0: CAN Bus error experienced
[   94.370747] mcp25xxfd spi0.0 can0: CAN Bus error experienced

Everything setup according to your instructions, terminators are turned on, power is applied via Raspberry PI's USB port.

Pillar1989, 2019-10-11:
@JuanAsturias Have you solved your problem ?

Pillar1989, 2019-10-28:
Closing for now. Happy to reopen if you have further questions.

closed 2019-10-28 by Pillar1989