Seeed-Studio/pi-hats #5: Your kernel headers for kernel 4.19.75-v8+ cannot be found at

CraftyMyner, 2019-10-25:
When running "install.sh" on RPi4, I get this error:

```
Error! echo
Your kernel headers for kernel 4.19.75-v8+ cannot be found at
/lib/modules/4.19.75-v8+/build or /lib/modules/4.19.75-v8+/source.
```

Full log:

```
pi@raspberrypi:~/Desktop/can/pi-hats/CAN-HAT $ sudo ./install.sh
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20190925+1-1).
The following NEW packages will be installed:
  raspberrypi-kernel-headers
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 24.9 MB of archives.
After this operation, 163 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20190925+1-1 [24.9 MB]
Fetched 24.9 MB in 13s (1,860 kB/s)
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 94163 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20190925+1-1) ...
Setting up raspberrypi-kernel-headers (1.20190925+1-1) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.20.1-2).
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  can-utils dkms
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 177 kB of archives.
After this operation, 715 kB of additional disk space will be used.
Get:1 http://mirrors.switch.ca/raspbian/raspbian buster/main armhf can-utils armhf 2018.02.0-1 [103 kB]
Get:2 http://mirrors.switch.ca/raspbian/raspbian buster/main armhf dkms all 2.6.1-4 [74.4 kB]
Fetched 177 kB in 1s (172 kB/s)
Selecting previously unselected package can-utils.
(Reading database ... 144752 files and directories currently installed.)
Preparing to unpack .../can-utils_2018.02.0-1_armhf.deb ...
Unpacking can-utils (2018.02.0-1) ...
Selecting previously unselected package dkms.
Preparing to unpack .../archives/dkms_2.6.1-4_all.deb ...
Unpacking dkms (2.6.1-4) ...
Setting up can-utils (2018.02.0-1) ...
Setting up dkms (2.6.1-4) ...
Processing triggers for man-db (2.8.5-2) ...

Creating symlink /var/lib/dkms/mcp2517fd/0.1/source ->
                 /usr/src/mcp2517fd-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.....
make -j4 KERNELRELEASE=4.19.75+ -C /lib/modules/4.19.75+/build M=/var/lib/dkms/mcp2517fd/0.1/build......
cleaning build area....

DKMS: build completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75-v7+ -C /lib/modules/4.19.75-v7+/build M=/var/lib/dkms/mcp2517fd/0.1/build......
cleaning build area...

DKMS: build completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75-v7l+ -C /lib/modules/4.19.75-v7l+/build M=/var/lib/dkms/mcp2517fd/0.1/build.....
cleaning build area...

DKMS: build completed.
Error! echo
Your kernel headers for kernel 4.19.75-v8+ cannot be found at
/lib/modules/4.19.75-v8+/build or /lib/modules/4.19.75-v8+/source.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

```

The CAN interface can not be found in "dmesg" or be brought up with "ip link"

Pillar1989, 2019-10-28:
@CraftyMyner We have updated the installation script and you can try again.

CraftyMyner, 2019-10-28:
Good to hear, I got it working Thanks!

Pillar1989, 2019-10-28:
@CraftyMyner  Thank you for your support and wish you good health and all the best

closed 2019-10-28 by Pillar1989