Seeed-Studio/pi-hats #6: CAN扩展卡install.sh安装完毕后导致树莓派黑屏

sume-cn, 2019-10-31:
树莓派3B和3B+均如此。
8、9月份的版本不会这样。

baorepo, 2019-10-31:
有添加其他的overlay吗？

sume-cn, 2019-10-31:
没有，提示下载的kernel版本20190925+1-1
your kernel version is
Not found **** coressponding *** kernel headers with apt-get
Choose *** y *** will revert the kernel to version 4.19.75+ then continue.
Choose *** n *** will exit without this driver support.
选择y安装重启后黑屏

sume-cn, 2019-10-31:
使用旧版（大约是8月份）的git备份安装，发现安装过程中依然下载的是20190925.
安装过程顺利，没有上述提示，但是重启后仍然提示failed to load kernel module，然后黑屏。

sume-cn, 2019-10-31:
请问你们能够复现吗？

sume-cn, 2019-10-31:
/etc/kernel/postinst.d/dkms:
Error! echo
Your kernel headers for kernel 4.19.75-v8+ cannot be found at
/lib/modules/4.19.75-v8+/build or /lib/modules/4.19.75-v8+/source.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.19.75-v8+ /boot/kernel8.img
--2019-10-31 12:47:28--  http://mirrors.ustc.edu.cn/archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb
Resolving mirrors.ustc.edu.cn (mirrors.ustc.edu.cn)... 218.104.71.170, 2001:da8:d800:95::110
Connecting to mirrors.ustc.edu.cn (mirrors.ustc.edu.cn)|218.104.71.170|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 24898072 (24M) [application/vnd.debian.binary-package]
Saving to: '/tmp/raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb'

/tmp/raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb    100%[=========================================================================================================================================>]  23.74M   939KB/s    in 25s     

2019-10-31 12:47:58 (986 KB/s) - '/tmp/raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb' saved [24898072/24898072]

(Reading database ... 206572 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20190925+1-1) over (1.20190925+1-1) ...
Setting up raspberrypi-kernel-headers (1.20190925+1-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75-v8+
Error! echo
Your kernel headers for kernel 4.19.75-v8+ cannot be found at
/lib/modules/4.19.75-v8+/build or /lib/modules/4.19.75-v8+/source.
./install.sh: line 35: get_kernel_version: command not found
 !!! Your kernel version is 
     Not found *** coressponding *** kernel headers with apt-get.
     This may occur if you have ran 'rpi-update'.
 Choose  *** y *** will revert the kernel to version 4.19.75+ then continue.
 Choose  *** N *** will exit without this driver support, by default.
Would you like to proceed? (y/N)

sume-cn, 2019-10-31:
现在的情况是，安装中提示：
Your kernel headers for kernel xxxxxxxxxxxx cannot be found at xxxxxxxxxxxxxxxx.
重启后提示failed to load kernel module，不黑屏了，但是dmesg | grep spi和ifconfig -a都找不到CAN设备。

sume-cn, 2019-11-01:
树莓派3B+，NOOBS 3.2.1(2019-09-30)，我再来试一下Raspbian Buster

sume-cn, 2019-11-01:
问题解决！参考这里：https://forum.seeedstudio.com/viewtopic.php?p=56399&sid=df8d3f4ad6b7f041f3be56eb55a77b63

6. Answer with yes: y
7. Installation will fail with
8. make
9. make install
10. sudo modprobe mcp25xxfd-can
11. sudo reboot

turmary, 2019-11-01:
We tried to fix this problem today and updated the installation script.

closed 2019-11-09 by sume-cn