Seeed-Studio/pi-hats #7: Heavy traffic will

sume-cn, 2019-11-09:


jenshz, 2019-11-11:
I'm having the same problem, but without heavy writes - It seems to happen after around 5-10 minutes of testing. The error reported is:
```
[ 1909.422685] mcp25xxfd spi0.0 can0: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
[ 1909.422758] mcp25xxfd spi0.0 can0: tefif: fifo 4 not pending - tef data: id: 0013d803 flags: 00000817, ts: 6991dd58 - this may be a problem with spi signal quality- try reducing spi-clock speed if this can get reproduced
[ 1914.423354] mcp25xxfd spi0.0 can0: tefif: fifo 1 not pending - tef data: id: 0013e803 flags: 00000217, ts: 6991eb0f - this may be a problem with spi signal quality- try reducing spi-clock speed if this can get reproduced
```
Restarting the interface (ip link down and then up) restores functionality for a similar duration, but eventually the same error as above happens.

sume-cn, 2019-11-12:
It seems to be the new driver's fault, in Aug and Sept I used to run my project with 100 Hz multiple 8 messages, it was good during 3 hour+ test. The hardware was same.
How can I roll back to the old driver?

jenshz, 2019-11-12:
I just tested with an older version of the kernel: `Linux raspberrypi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux` from the previous release of Raspbian, but the problem is also present there. I think it might be some kind of bug in the `mcp25xxfd` driver itself. Bringing the network interface (can0 / can1) down and then up is a workaround for now, but not ideal.

sume-cn, 2019-11-13:
I've tested another board mounting MCP2515(driver version 2017) with 1300 fps message rate, 500kbps baud rate, for 17 hours, and it's OK.

jenshz, 2019-11-14:
I've tried the driver here: https://github.com/GBert/misc/tree/master/RPi-MCP2517 (changing the oscillator setting to 40 MHz), and that worked for around 15 hours before failing hard. After that, nothing but a full power cycle would bring it online again. At least the driver in this repository works after a software reset, if only for a short while.

johandc, 2019-11-15:
Same issue with the CAN-HAT MCP2517 on latest raspbian 4.19.75-v7l, on Raspberry Pi4. It works for a while, and the suddenly stops. It seems the higher the load, the higher likelihood of getting the error.

johandc, 2019-11-15:
It seems related to the fact that the MCP2517FD have now been discontinued due to a quite severe silicon bug, described in this errata document:
http://ww1.microchip.com/downloads/en/DeviceDoc/MCP2517FD-Data-Sheet-Errata-DS80000792B.pdf

They specifically mention this to be related to slower linux (raspberry pi) and a too long delay between spi bytes:

```
TX MAB underflow/RX MAB overflow due to
long delays between SPI bytes:

The SPI Interface can block the CAN FD Controller
module from accessing RAM in between SPI bytes
and between the last byte and the rising edge of
the nCS line during an SPI READ or SPI READ
CRC instruction while accessing RAM.
If the CAN FD Controller module is blocked for
more than TSPIMAXDLY, a TX MAB underflow or an
RX MAB overflow can occur.

Work around:

Keep the delay between two SPI bytes and
between the last SPI byte and the rising edge of
nCS shorter than TSPIMAXDLY; see Figure 1.
The maximum allowed delay between two bytes
depends on which CAN message frame is
transmitted and on the selected Nominal Bit Time
(NBT) and Data Bit Time (DBT). Table 1 lists
TSPIMAXDLY for the worst-case scenarios.
For example: TSPIMAXDLY is 8.5 µs for a CAN FD
frame at 500 kbps/2 Mbps. In comparison, an SPI
byte takes 0.67 µs at 12 MHz SCK. A delay of 10
times the duration of one SPI byte could cause a
TX MAB underflow. It is highly unlikely for an MCU
application to introduce such a long delay, but this
error could occur when running an operation
system like Linux on a slower MPU.
In case of a TX MAB underflow, the device will
notify the application by setting SERRIF and
MODIF and by transitioning to Restricted
Operation or Listen Only mode (depending on
CiCON.SERR2LOM). After the application
requests Normal mode, the CAN FD Controller
module will automatically attempt to retransmit the
message that caused the TX MAB underflow. It is
not necessary to reset the device.
In case of an RX MAB overflow, the device will
notify the application by setting SERRIF. The
device will remain in Normal mode. The message
that caused the RX MAB overflow will be
discarded.
```

johandc, 2019-11-19:
We tested with a board using the MCP251**8**FD where the errata of the MCP2517 have been fixed. This did not solve the issue with the "Something went wrong - we got a TEF..." error.

jorisa, 2019-12-04:
Note, this appears to be  this issue: https://github.com/msperl/linux-rpi/issues/6

Pillar1989, 2019-12-06:
We will also upgrade this product to ***MCP2518FD*** in the future

jbakuwel, 2020-01-12:
I'm bitten by this too, a real show stopper. @Pillar1989 do you know who might resolve the PI kernel bug (if that is indeed the cause of this issue and more importantly do you know if whoever might resolve the PI kernel bug is aware of the bug?

Pillar1989, 2020-01-14:
@jbakuwel  https://github.com/msperl/linux-rpi/issues/6  Here are some updates; I don't know if they are what you want. Besides, we are experiencing the Chinese New Year, and we will not have enough energy to devote to this matter until February 1, 2020.

jbakuwel, 2020-01-14:
> @jbakuwel msperl/linux-rpi#6 Here are some updates; I don't know if they are what you want. Besides, we are experiencing the Chinese New Year, and we will not have enough energy to devote to this matter until February 1, 2020.

Hi Baozhu,

Thanks for the prompt response. I did read all the posts on github and even tried reversing the TEf optimisations as some people seem to think these might have introduced the problem but so far have no luck getting the hardware to behave.

Wish you a great festive season and hope to hear from you re this in Feb.

kind regards,
Jan

jbakuwel, 2020-01-26:
Reverting the commits as suggested by @irrwisch1 [here](https://github.com/msperl/linux-rpi/issues/6) makes seems to resolve the problem for me too on a Raspberry Pi 4. It's more or less stable after changing MCP25XXFD_SCLK_DIVIDER to 4 in mcp25xxfd_regs.h

However I do see these appear in syslog as well:

test kernel: [ 1538.699553] mcp25xxfd spi1.0: ECC double bit error at c00
test kernel: [ 1538.699695] mcp25xxfd spi1.0: unidentified system interrupt - intf =  391a1108
test kernel: [ 1538.699846] mcp25xxfd spi1.0: ECC double bit error at c00
test kernel: [ 2222.497103] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a1118
test kernel: [ 2222.497225] mcp25xxfd spi1.0: ECC double bit error at 800

This problem was resolved by applying the changes suggested by @jrm06c [here](https://github.com/msperl/linux-rpi/issues/8)

test kernel: [ 1913.359886] mcp25xxfd spi0.0: found IVMIF situation not supported by driver - bdiag = [0x00008400, 0x00005ed7]

It would be great to get this driver stable; happy to help in anyway I can.

jbakuwel, 2020-01-26:
Further debug info. I've written a simple program that forwards packets from can0 (called b) to can1 (called v) and vice versa. This runs fine for about 7 minutes (having processed ~70K frames), then single bit ECC errors are showing up. These are (most likely) corrected and all continues to run fine.
Then after ~150K frames, I start to see double ECC errors with subsequent transmit errors.

See my previous post for the changes I made to the driver.

Any suggestions?

20200126-142108: Frames received on can0 (b) will be forwarded to can1 (v)
20200126-142108: Frames received on can1 (v) will be forwarded to can0 (b)
20200126-142108: Starting notifiers
20200126-142108: Starting test
20200126-142142: Processing v queue: 25 messages
20200126-142142: Processing b queue: 8 messages
20200126-142336: Saving statistics to /data/logs/20200126-142336-test-v-stats.log
20200126-142336: Saving statistics to /data/logs/20200126-142336-test-b-stats.log
20200126-142336: Processing v queue: 24579 messages
20200126-142340: Processing b queue: 2493 messages
20200126-142707: Saving statistics to /data/logs/20200126-142707-test-v-stats.log
20200126-142707: Saving statistics to /data/logs/20200126-142707-test-b-stats.log
20200126-142707: Processing v queue: 44220 messages
20200126-142717: Processing b queue: 6007 messages
kernel: [ 7050.997314] mcp25xxfd spi1.0: ECC single bit error at c00
20200126-143028: Saving statistics to /data/logs/20200126-143028-test-v-stats.log
20200126-143028: Saving statistics to /data/logs/20200126-143028-test-b-stats.log
20200126-143028: Processing v queue: 40758 messages
20200126-143029: Processing b queue: 914 messages
kernel: [ 7297.112660] mcp25xxfd spi1.0: ECC single bit error at 800
kernel: [ 7304.042530] mcp25xxfd spi1.0: ECC single bit error at 800
20200126-143404: Saving statistics to /data/logs/20200126-143404-test-v-stats.log
20200126-143404: Saving statistics to /data/logs/20200126-143404-test-b-stats.log
20200126-143404: Processing v queue: 45941 messages
20200126-143405: Processing b queue: 643 messages
kernel: [ 7446.778260] mcp25xxfd spi1.0: ECC double bit error at c00
kernel: [ 7446.778557] mcp25xxfd spi1.0: ECC double bit error at c00
20200126-143504: b->v: CanError('Transmit buffer full')

marckleinebudde, 2020-01-28:
@jbakuwel can you send me your test program?

jbakuwel, 2020-01-28:
@marckleinebudde, yeah sure no worries: [cantxrx.py.txt](https://github.com/Seeed-Studio/pi-hats/files/4125121/cantxrx.py.txt). Note tabstop=3 makes for easy reading.
I see you're working with Victron equipment too :-) ?

marckleinebudde, 2020-01-29:
@jbakuwel I'm on a rapi3, using a [PiCAN FD Duo](http://skpang.co.uk/catalog/pican-fd-duo-board-with-real-time-clock-for-raspberry-pi-3-p-1568.html), which has two `mcp2517fd`, both connected to `spi0` and is probably similar to the Seed Studio's. Both lines are connected and terminated.

- What bitrate are you using in your test?
- How do you inject CAN messages into the loop?

Can you improve your python test, that it doesn't loopback the error frames that are received on a interface:
```
 (2020-01-29 16:46:00.992598)  can0  RX - -  002   [1]  00                                                                                                                                                                                                                                
 (2020-01-29 16:46:00.993560)  can1  RX - -  20000004   [8]  00 08 00 00 00 00 00 00   ERRORFRAME                                                                                                                                                                                         
        controller-problem{tx-error-warning}                                                                                                                                                                                                                                              
 (2020-01-29 16:46:00.993705)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 00 00   ERRORFRAME                                                                                                                                                                                         
        controller-problem{tx-error-warning}                                                                                                                                                                                                                                              
 (2020-01-29 16:46:00.993868)  can1  RX - -  20000004   [8]  00 20 00 00 00 00 00 00   ERRORFRAME                                                                                                                                                                                         
        controller-problem{tx-error-passive}                                                                                                                                                                                                                                              
 (2020-01-29 16:46:00.993899)  can0  RX - -  20000004   [8]  00 20 00 00 00 00 00 00   ERRORFRAME                                                                                                                                                                                         
        controller-problem{tx-error-passive}                                                                                                                                                                                                                                              
 (2020-01-29 16:46:00.994153)  can0  RX - -  002   [1]  08                                                                                                                                                                                                                                
 (2020-01-29 16:46:00.994184)  can1  TX - -  002   [1]  08                                                                                                                                                                                                                                
[...]
 (2020-01-29 16:46:01.001755)  can0  TX - -  002   [1]  08
 (2020-01-29 16:46:01.002159)  can1  RX - -  002   [1]  09
 (2020-01-29 16:46:01.002306)  can0  TX - -  002   [1]  09
 (2020-01-29 16:46:01.002798)  can1  RX - -  004   [8]  00 08 00 00 00 00 00 00                                                              
```

jbakuwel, 2020-01-29:
@marckleinebudde I'm using a bitrate of 500kbps. I've been doing two tests. For the first test I connect can0 to can1 with two wires and use cangen and candump. This test appears to run fine, at least for a day or so. Hence the need for another test that reliably reproduces the issue(s).

This other test uses my Python script on a live CAN bus with plenty of traffic (@ 500kbps) that normally connects two nodes. For my test, both nodes are connected to the Pi HAT instead of directly to each other. The Pi running my Python script functions as the wire mentioned in the first test. If this can run for an extended period of time without causing issues (either showing up in the Pi kernel log or my Python script) or in the live CAN system, we can hopefully conclude that the issues with the driver have been resolved.

I've added a test for error frames in the code although I do not see them appear (ie. makes no difference). Maybe these are not passed on by the Python CAN library.

The Seeed Studio Pi Hat also has two mcp2517fd's, but those are connected to spi0 and spi1.

After a while (sometimes 15 minutes, sometimes an hour), I see:

[47007.453669] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a1118
[47007.453792] mcp25xxfd spi1.0: ECC double bit error at 800
20200130-113800: b->v: CanError('Transmit buffer full')

This suggests a timing issue / bug in the driver causing a data corruption.

from which the driver does not recover. I can't restart it either:

# ip link set can1 type can restart
RTNETLINK answers: Invalid argument

but have to reload the kernel module.

I'm bringing the devices up at boot with:

set_can () {
	ip link set $1 up type can bitrate 500000 restart-ms 100 fd off
	ifconfig $1 txqueuelen 65536
	ifconfig $1 up
}

jbakuwel, 2020-02-11:
Hi folks,

Back at testing the driver after having to rebuild the test system due to a failing SD card.
I've started at the beginning; ie with the driver as currently published by Seeed Studio.
Now getting a slightly different message after [this](https://github.com/Seeed-Studio/pi-hats/commit/0db233c054cdd1b37fe7d6f6c3c6ecd019742449) commit.

Kernel log filling up with heaps of these (possibly one message for each forwarded frame while running my 2nd test see above):

Feb 12 08:40:28 mitm kernel: [43531.353724] mcp25xxfd spi0.0 can0: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
Feb 12 08:40:28 mitm kernel: [43531.353831] mcp25xxfd spi1.0 can1: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
Feb 12 08:40:28 mitm kernel: [43531.354085] mcp25xxfd spi1.0 can1: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
Feb 12 08:40:28 mitm kernel: [43531.354318] mcp25xxfd spi1.0 can1: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
Feb 12 08:40:28 mitm kernel: [43531.354561] mcp25xxfd spi1.0 can1: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
Feb 12 08:40:28 mitm kernel: [43531.354695] mcp25xxfd spi0.0 can0: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo

marckleinebudde, 2020-02-11:
Hey @jbakuwel, good to see you back!
In the mean time, I've squashed my driver into a nice patch stack, which is available here:
https://github.com/marckleinebudde/linux/tree/v4.19-rpi/mcp25xxfd-20200211-2
I'd really appreciate if you give it a go.

jbakuwel, 2020-02-11:
Hey @marckleinebudde, glad to be back :-)
Still very busy with other stuff so working on this in between other things. Do you want me to test the complete kernel with driver from your repo or just the driver, ie. are there changes to other parts of the kernel that are relevant as well?

jbakuwel, 2020-02-11:
Interesting point to note: despite the kernel messages mentioned above:
Feb 12 08:40:28 mitm kernel: [43531.353724] mcp25xxfd spi0.0 can0: Something is wrong - we got a TEF interrupt but we were not able to detect a finished fifo
the live CAN system (so far) seems to be happy. I'll let it run and expect to see ECC errors showing up after a while; after all nothing has changed yet in that department.

marckleinebudde, 2020-02-11:
@jbakuwel The driver is based on the latest v4.19-rpi kernel and includes some backports for the spi-aux driver. Using the latest stable kernel is always a good idea. So compile the whole kernel.

jbakuwel, 2020-02-12:
Update (still on current Seeed Studio driver and 4.19.97-v7l+ kernel):

Feb 12 09:16:13 test kernel: [45675.865861] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:03:43 test kernel: [59326.835049] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 13:03:43 test kernel: [59326.835426] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 3 to 6
Feb 12 13:14:35 test kernel: [59978.328018] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:18:49 test kernel: [60232.023712] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:19:32 test kernel: [60275.729571] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:20:58 test kernel: [60361.052802] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:21:16 test kernel: [60379.242129] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:26:37 test kernel: [60700.705055] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 13:26:37 test kernel: [60700.705152] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 3 to 6
Feb 12 13:32:31 test kernel: [61054.898096] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 13:32:31 test kernel: [61054.898234] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 3 to 6

and after increasing the traffic a bit:
Feb 12 13:53:01 test kernel: [62284.237189] mcp25xxfd spi1.0: ECC single bit error at 000

Otherwise everything still running...

jbakuwel, 2020-02-12:
As expected, the driver didn't recover after the first double bit ECC error. The "mode switches" are new... haven't seen those before. Kernel 4.19.97-v7l+ with current Seeed Studio driver.

@marckleinebudde, will now compile your kernel + driver and give that a go.

Feb 12 09:16:13 test kernel: [45675.865861] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:03:43 test kernel: [59326.835049] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 13:03:43 test kernel: [59326.835426] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 3 to 6
Feb 12 13:14:35 test kernel: [59978.328018] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:18:49 test kernel: [60232.023712] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:19:32 test kernel: [60275.729571] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:20:58 test kernel: [60361.052802] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:21:16 test kernel: [60379.242129] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 7 to 6
Feb 12 13:26:37 test kernel: [60700.705055] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 13:26:37 test kernel: [60700.705152] mcp25xxfd spi0.0: Controller unexpectedly switched from mode 3 to 6
Feb 12 13:32:31 test kernel: [61054.898096] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 13:32:31 test kernel: [61054.898234] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 3 to 6
Feb 12 13:53:01 test kernel: [62284.237189] mcp25xxfd spi1.0: ECC single bit error at 000

Feb 12 15:00:23 test kernel: [66326.870662] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.870799] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a1108
Feb 12 15:00:23 test kernel: [66326.870819] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 6 to 3
Feb 12 15:00:23 test kernel: [66326.870839] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.870979] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.871023] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.871043] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.871186] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.871230] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.871250] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.871389] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.871433] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.871452] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.871591] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.871635] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.871655] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.871793] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.871837] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.871857] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.871996] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.872040] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.872059] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.872198] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.872242] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.872261] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.872402] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.872446] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.872466] mcp25xxfd spi1.0: ECC double bit error at 000
Feb 12 15:00:23 test kernel: [66326.872605] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.872650] mcp25xxfd spi1.0: unidentified system interrupt - intf =  b91a9100
Feb 12 15:00:23 test kernel: [66326.872805] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.872997] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.873190] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.873382] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.873574] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.873766] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.873958] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.874276] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3
Feb 12 15:00:23 test kernel: [66326.874603] mcp25xxfd spi1.0: Controller unexpectedly switched from mode 7 to 3

marckleinebudde, 2020-02-12:
@jbakuwel Due to the [errata](http://ww1.microchip.com/downloads/en/DeviceDoc/MCP2517FD-Silicon-Errata-and-Data-Sheet-Clarification-DS80000792A.pdf) the mcp2517fd can experience TX MAB underflows. With this error it will switch from CAN-2.0 Mode (6) to Restricted Mode (7). Then the driver brings the device back into CAN-2.0 Mode. However from the driver's point of view the transition from Restricted to CAN-2.0 (6 -> 7) is unexpected.

For yet unknown reasons, in contradiction to the errata the devive will sometimes switch to Listen Only Mode (3) instead of Restricted. This (6 -> 3) and the recovery (3 -> 6) are also not expected by the driver.

A debug version of my driver read the mode value several times and it seems the device sometimes changes first into Listen Only and then into Restricted:
`CAN-2.0 ---<TX MAB Underflow>---> Listen Only --> Restricted`
I don't know if the Listen Only -> Restricted change creates an interrupt. 

I cannot explain the 7 -> 3 transition. Might be a recovery 7 -> 6, followed immediately by a TX MAB error, where the device changes into Listen Only. The interrupt handler's delay might be so high, that the device already changed to Listen Only.

Long story short, the driver doesn't take all these mode changes into account.
```
#define MCP25XXFD_CAN_CON_MODE_LISTENONLY 3
#define MCP25XXFD_CAN_CON_MODE_CAN2_0 6
#define MCP25XXFD_CAN_CON_MODE_RESTRICTED 7

```

jbakuwel, 2020-02-12:
@marckleinebudde I wonder why I haven't seen those mode switches before in all my tests... the patch authored by @jenshz should have no effect on this. Or maybe it does...

I have compiled your kernel and driver - I presume. This is what I did:

- git clone --depth=1 --branch v4.19-rpi/mcp25xxfd-20200211-2 https://github.com/marckleinebudde/linux.git
- KERNEL=kernel7l
- make bcm2711_defconfig
- make -j4 zImage modules dtbs
- sudo make modules_install
- sudo cp arch/arm/boot/dts/*.dtb /boot/
- sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
- sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
- sudo cp arch/arm/boot/zImage /boot/$KERNEL.img
- reboot
- uname -a
Linux test 4.19.102-v7l+ #1 SMP Wed Feb 12 18:16:40 NZDT 2020 armv7l GNU/Linux

test@test: $ sudo ./reload-interfaces.sh
rmmod: ERROR: Module mcp25xxfd_can is not currently loaded
modprobe: FATAL: Module mcp25xxfd_can not found in directory /lib/modules/4.19.102-v7l+
Cannot find device "can0"
Cannot find device "can0"
Cannot find device "can0"
Cannot find device "can1"
Cannot find device "can1"
Cannot find device "can1"
test@test: $ find /lib/modules/4.19.102-v7l+/ -iname 'mcp*' -print
/lib/modules/4.19.102-v7l+/kernel/drivers/net/can/spi/mcp251x.ko
/lib/modules/4.19.102-v7l+/kernel/drivers/iio/adc/mcp3422.ko
/lib/modules/4.19.102-v7l+/kernel/drivers/iio/adc/mcp320x.ko

It's getting late here and brain refuses to dig deeper into this after a long day :-)
What am I missing?

marckleinebudde, 2020-02-12:
> make bcm2711_defconfig

I should have activated the driver in the defconfigs....For now switch it on manually:
```
make menuconfig
    -> Networking support
    -> CAN bus subsystem support
    -> CAN Device Drivers
    -> CAN SPI interfaces
    -> Microchip MCP25xxFD SPI CAN controllers
        -> Additional Sanity Checks
        -> Logging
```
> make -j4 zImage modules dtbs

marckleinebudde, 2020-02-12:
I think the seedstudio's overlay should work with my driver, but I haven't tested it. Post the output of `dtc -I fs /proc/device-tree` if the driver doesn't find the device.

I'm using this snippet in my `/boot/config.txt`:
```
# Seedstudio CAN-BUS FD
dtoverlay=mcp25xxfd-spi0-0,interrupt=25
dtoverlay=mcp25xxfd-spi1-0,interrupt=24
```

jbakuwel, 2020-02-12:
@marckleinebudde Thanks for helping my poor brain :-) I'll start my test and will report back to you tomorrow morning my time, tonight your time.  The standard overlay works fine.

jbakuwel, 2020-02-12:
@marckleinebudde Test running. I can report this at this stage:
Feb 12 21:30:13 test kernel: [ 2517.408097] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.

marckleinebudde, 2020-02-12:
This means at least one lost CAN frame during RX. In CAN-2.0 mode the RX FIFO holds 32 CAN frames. But I've plans to support a second FIFO. Have you decreased the `spimaxfrequency` via `config.txt`?

jbakuwel, 2020-02-12:
@marckleinebudde A bit more info from the log (omitting more RX-FIFO overflow in FIFO 2 messages; these happen for both spi1.0 (less frequent) and spi0.0 (more frequent)):

Feb 12 21:39:56 test kernel: [ 3099.960780] mcp25xxfd spi1.0 can1: Stopping tx-queue (tx_head=0x00057754, tx_tail=0x0005774c, len=8).
Feb 12 22:10:13 test kernel: [ 4917.527332] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x04c8.

I have not decreased spimaxfrequency in config.txt; default Raspbian system running 4.19.102-v7l+.

I expect the double bit ECC error will show up at some time... but for now the live CAN system is happy (ie. hasn't yet detected a fault with the data).

jbakuwel, 2020-02-12:
And here's the config for the interfaces:

#!/bin/bash

set_can () {
	ip link set $1 type can bitrate 500000 restart-ms 100 fd off
	ip link set $1 txqueuelen 65535
	ip link set $1 up
}

rmmod mcp25xxfd_can
modprobe mcp25xxfd_can

set_can can0
set_can can1

marckleinebudde, 2020-02-12:
There is a race condition in the driver, TX-Complete vs. TX wakeup, fixing this properly costs 20% of performance on my rpi1, working around is much cheaper. When this happens `Stopping tx-queue` is printed. Should document this in the driver properly and make it a `netdev_dbg()`.

marckleinebudde, 2020-02-12:
The interface config looks good.

jbakuwel, 2020-02-12:
Running on a RPi4 here.

marckleinebudde, 2020-02-12:
I haven't seen any RX-Overflows on my rpi3. Should test on rpi4.

jbakuwel, 2020-02-12:
Just a sample from the log:

Feb 12 22:07:43 test kernel: [ 4767.274098] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:08:26 test kernel: [ 4810.320286] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:09:08 test kernel: [ 4852.307645] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:09:19 test kernel: [ 4863.316216] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:11:05 test kernel: [ 4969.581118] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:11:07 test kernel: [ 4971.384544] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:11:25 test kernel: [ 4989.101337] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:14:09 test kernel: [ 5153.567410] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:15:09 test kernel: [ 5213.427739] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:15:41 test kernel: [ 5245.817137] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:17:10 test kernel: [ 5334.505669] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:17:33 test kernel: [ 5357.172694] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:17:50 test kernel: [ 5374.217732] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:19:14 test kernel: [ 5458.775201] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:20:32 test kernel: [ 5536.241862] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:20:34 test kernel: [ 5538.943836] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:20:35 test kernel: [ 5539.743918] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:23:24 test kernel: [ 5708.776282] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:23:59 test kernel: [ 5743.762741] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:25:21 test kernel: [ 5825.246199] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:25:21 test kernel: [ 5825.545287] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:26:45 test kernel: [ 5909.529351] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:26:48 test kernel: [ 5912.633137] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:29:00 test kernel: [ 6044.136954] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:29:10 test kernel: [ 6054.345870] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 12 22:29:28 test kernel: [ 6072.093586] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 12 22:31:29 test kernel: [ 6193.554320] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.

marckleinebudde, 2020-02-12:
Oh, that's quite often :(

jbakuwel, 2020-02-12:
I think the driver now lost the plot, the following message is repeated many times per sec. Bedtime for me, have a good day :-)

Feb 12 22:36:46 test kernel: [ 6510.614345] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.614587] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.614758] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.614860] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.614961] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615063] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615164] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615266] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615367] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615475] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615725] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615838] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.615945] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.
Feb 12 22:36:46 test kernel: [ 6510.616081] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0488.

jbakuwel, 2020-02-12:
Oh, and the system just crashed as well. I stopped the Python script but the messages in the log kept coming; maybe these were buffered but it could also be that the driver got itself in a spin from which it would not recover.

marckleinebudde, 2020-02-12:
I test on my rpi4 and come back to you. gn8

jbakuwel, 2020-02-23:
@marckleinebudde Just wondering how it's going. Anything I can do to help?

marckleinebudde, 2020-02-24:
While working with a rpi3 I figured out that sometimes the 1st byte or the CAN-ID of CAN frames read from the controller has a bit flip. Here the CAN-ID should be `0x77`, but it was `0xf7`:
```
Feb 22 10:01:51 rpi3 kernel: mcp25xxfd spi1.0 can1: RX-0: offset=7, len=1, object=7/0x07, canid=000000f7
Feb 22 10:01:51 rpi3 kernel:             data = f7 00 00 00  08 00 00 00
Feb 22 10:01:51 rpi3 kernel:                    0d 92 91 c9  c7 c8 c9 ca
Feb 22 10:01:51 rpi3 kernel:                    cb cc cd ce
```

Here the 1st data byte should be `0xf5` but it was `0x75`
```
Feb 23 19:49:16 rpi3 kernel: mcp25xxfd spi0.0 can0: RX-0: data expected=f5, got=75
Feb 23 19:49:16 rpi3 kernel:             data = 78 00 00 00  08 00 00 00
Feb 23 19:49:16 rpi3 kernel:                    3f ef 7f ca  75 f6 f7 f8
Feb 23 19:49:16 rpi3 kernel:                    f9 fa fb fc
```

I eliminated that class of errors by using the CRC transfer feature on the mcp2517fd. In the above mentioned cases the CRC is broken and the transfer is retried. Sometimes the CRC is broken, but the CAN-ID and the data looks good, this seems to be the problem that is described in the mcp2517fd's errata sheet.

Further the driver now checks if the thread associated with the SPI Controller is scheduled with real time priorities - usually it isn't. The driver will print a message to the logs in the form of (where PID is replaced by the actual PID of the thread):
```
SPI thread not scheduled with RT priority, expect poor performance. Reconfigure with: 'chrt -f -p 1 <PID>'",
```
To make this permanent add this to your startup scripts:
```
chrt -f -p 1 `pidof spi0`
chrt -f -p 1 `pidof spi1`
```
Next I'll update the TX-PATH to use CRC transfers as well, maybe this helps even more.

Can you give it another short test. The current driver is available in `mcp25xxfd-20200211-3` branch:
https://github.com/marckleinebudde/linux

jbakuwel, 2020-02-28:
@marckleinebudde Thanks for that. Test is running, so far so good (as in driver continues to run, live CAN system happy, no ECC errors). But:

Feb 28 19:21:06 test kernel: [ 1470.418732] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:08 test kernel: [ 1472.520786] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:09 test kernel: [ 1473.020464] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:14 test kernel: [ 1478.525127] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:20 test kernel: [ 1483.631749] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:25 test kernel: [ 1489.537997] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:33 test kernel: [ 1496.753161] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:46 test kernel: [ 1509.584868] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:46 test kernel: [ 1510.185041] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:49 test kernel: [ 1512.591132] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:54 test kernel: [ 1517.791889] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:55 test kernel: [ 1518.692150] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:21:59 test kernel: [ 1522.800577] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:00 test kernel: [ 1524.400587] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:22 test kernel: [ 1546.028679] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:28 test kernel: [ 1552.538172] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:33 test kernel: [ 1557.458603] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:37 test kernel: [ 1561.461031] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:56 test kernel: [ 1579.784572] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:22:56 test kernel: [ 1580.186095] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:02 test kernel: [ 1586.392456] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:11 test kernel: [ 1595.404363] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:21 test kernel: [ 1604.720108] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:52 test kernel: [ 1636.081787] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:53 test kernel: [ 1636.784686] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:55 test kernel: [ 1638.888977] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:23:59 test kernel: [ 1643.393622] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:02 test kernel: [ 1646.497012] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:06 test kernel: [ 1650.499197] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:08 test kernel: [ 1652.202024] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:16 test kernel: [ 1660.008380] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:18 test kernel: [ 1662.008685] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:18 test kernel: [ 1662.210749] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:26 test kernel: [ 1669.619292] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:42 test kernel: [ 1686.258962] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:47 test kernel: [ 1690.663138] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:48 test kernel: [ 1691.563521] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:48 test kernel: [ 1692.164678] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:24:52 test kernel: [ 1696.464965] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:25:00 test kernel: [ 1704.172320] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:25:11 test kernel: [ 1714.780924] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:25:12 test kernel: [ 1716.482184] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:25:12 test kernel: [ 1716.482276] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:25:40 test kernel: [ 1744.129699] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:25:48 test kernel: [ 1752.146231] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:02 test kernel: [ 1765.766387] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:02 test kernel: [ 1766.168039] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:32 test kernel: [ 1796.215899] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:39 test kernel: [ 1803.436067] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:40 test kernel: [ 1803.638517] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:52 test kernel: [ 1815.759818] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:56 test kernel: [ 1820.463122] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:26:57 test kernel: [ 1820.963154] mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2.
Feb 28 19:27:01 test kernel: [ 1825.465106] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:27:10 test kernel: [ 1834.473137] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
Feb 28 19:27:11 test kernel: [ 1834.673665] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.
[... cut ...]
Feb 28 19:29:38 test /testd: 20200228-192938: Processing b queue: 13037 messages
Feb 28 19:29:38 test /testd: 20200228-192938: Processing v queue: 33935 messages
Feb 28 19:30:39 test /testd: 20200228-193039: Processing b queue: 13038 messages
Feb 28 19:30:39 test /testd: 20200228-193039: Processing v queue: 33636 messages
Feb 28 19:31:40 test /testd: 20200228-193140: Processing b queue: 13035 messages
Feb 28 19:31:41 test /testd: 20200228-193141: Processing v queue: 33623 messages

I'll let it run and will report back later with more.

jbakuwel, 2020-02-28:
@marckleinebudde I read somewhere that the (Seeed Studio) driver can take a parameter to increase the number of FIFOs (but couldn't find the syntax of that parameter). Is that something worth trying? Which parameter would I pass to modprobe to do this for your driver?

jbakuwel, 2020-02-28:
@marckleinebudde Driver crashed the system after 391825 of these in just over two minutes:

First one:
Feb 28 19:49:23 test kernel: [ 3167.256306] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0478.

Last one:
Feb 28 19:51:36 test kernel: [ 3299.919855] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0478.

System crashed.

marckleinebudde, 2020-02-28:
Hello @jbakuwel,

have you configured the SPI thread priorities as mentioned in my last message?

> @marckleinebudde Thanks for that. Test is running, so far so good (as in driver continues to run, live CAN system happy, no ECC errors). But:
> 
> Feb 28 19:21:06 test kernel: [ 1470.418732] mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2.

In the mean time I figured out, that although the RX-Overflow bit is set in the controller, it's a RX-MAB overflow (which is caused by the errata), there's enough space in the FIFO.

> And to give you an idea of the number of messages being forwarded backwards and forwards:

Okay, quite busy bus :)

> @marckleinebudde I read somewhere that the (Seeed Studio) driver can take a parameter to increase the number of FIFOs (but couldn't find the syntax of that parameter). Is that something worth trying? Which parameter would I pass to modprobe to do this for your driver?

My driver uses up to 32 RX Objects, that is one complete FIFO. But the error message is misleading. See above. The current driver has that message fixed.

> Driver crashed the system after 391825 of these in just over two minutes:
>
> First one:
> ```Feb 28 19:49:23 test kernel: [ 3167.256306] mcp25xxfd spi1.0 can1: Single ECC Error corrected at address [...]

You mean 391825 of these error messages? Is the CAN adpater (`can1`) and the address (`0x0478`) the same for all these messages?

> System crashed.

How exactly? Just locked up? Or any other error message?

marckleinebudde, 2020-02-28:
@jbakuwel 

The next iteration of the driver: https://github.com/marckleinebudde/linux/tree/v4.19-rpi/mcp25xxfd-20200211-4

jbakuwel, 2020-02-28:
@marckleinebudde Yes, I did set the SPI priorities as mentioned in your message. All 391825 messages are the same (ie. can1 and 0x0478). It looks like the driver gets stuck processing the same frame as the log messages are being generated at a very high frequency until something somewhere gets exhausted. I am not sure (I don't have a serial console connected to the Pi) what exactly happened on the RPi. My ssh sessions terminated (as if I would have logged off) and the RPi no longer responded to pings.

I'll run another test with -4 and report back.

Do you think it's somehow possible to work around the RX-MAB overflow or should we consider the MCP2517FD not fit for purpose?

jbakuwel, 2020-02-28:
@marckleinebudde With -4, the repeated error message (1237517 in 3 mins 40 secs) in the log is: mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x04a4". I also continue to see "mcp25xxfd spi0.0 can0: RX-FIFO overflow in FIFO 2" and "mcp25xxfd spi1.0 can1: RX-FIFO overflow in FIFO 2" at irregular intervals varying from 1 to 12 seconds.

The return error code is 0 for both single and double bit ECC errors. I assume  mcp25xxfd_handle_eccif is repeatedly called for the same CAN frame but why and from where eludes me at the moment. The driver seizes to transmit frames once these (single bit) ECC errors are happening. But we should expect the driver to continue to transmit frames, right? Does the silicon perhaps turn the bus off after a single (or double bit for that matter) ECC error? The interface is configured with "restart-ms 100" but that might not be happening. Or perhaps once a ECC error occurs, the hardware continues to think there are ECC errors for subsequent CAN frames even if that's not the case?

Would it be possible for debugging purposes to print the contents of the offending CAN frame to syslog?

I also observe that I do not see mode changes with your driver.

The system did not crash this time. I did kill the CANTXRX process (which didn't stop the flooding of syslog but that could be a tail -f /var/log/syslog backlog), then I remove and reloaded the modules, which stopped the flood.

jbakuwel, 2020-02-28:
@marckleinebudde

> My driver uses up to 32 RX Objects, that is one complete FIFO. But the error message is misleading. See above. The current driver has that message fixed.

 but the message I see is still the same. Does that mean we're not hitting the RX-MAB overflow?

jbakuwel, 2020-02-29:
Ran another test without flipping the bits of ecc_stat. I know it's wild guess ;-)
But any easy test (I changed line 1547 from "~ecc_stat" to "ecc_stat"):

`err = regmap_update_bits(priv->map, MCP25XXFD_ECCSTAT,                                MCP25XXFD_ECCSTAT_IF_MASK, ~ecc_stat);
`

`  err = regmap_update_bits(priv->map, MCP25XXFD_ECCSTAT, MCP25XXFD_ECCSTAT_IF_MASK,  ecc_stat);
`

Result: 730691 times "mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x0478" in the logs (and well as the usual RX-FIFO overflow in FIFO 2".

marckleinebudde, 2020-02-29:
Hello @jbakuwel,

> With -4, the repeated error message (1237517 in 3 mins 40 secs) in the log is: mcp25xxfd spi1.0 can1: Single ECC Error corrected at address 0x04a4". I also continue to see "mcp25xxfd spi0.0 can0: [...]

Are you sure, you're using the correct version of the driver? In https://github.com/marckleinebudde/linux/tree/v4.19-rpi/mcp25xxfd-20200211-4, I've clarified the text in the error message.

https://github.com/marckleinebudde/linux/blob/v4.19-rpi/mcp25xxfd-20200211-4/drivers/net/can/spi/mcp25xxfd/mcp25xxfd-core.c#L1336 :
```
	/* If SERRIF is active, there was a RX MAB overflow. */
	if (priv->regs_status.intf & MCP25XXFD_CAN_INT_SERRIF) {
		if (priv->devtype_data->quirks &
		    MCP25XXFD_QUIRK_MAB_ERROR_NO_WARN)
			netdev_dbg(priv->ndev,
				   "RX-%d: MAB overflow detected.\n",
				   ring->nr);
		else
			netdev_info(priv->ndev,
				    "RX-%d: MAB overflow detected.\n",
				    ring->nr);
	} else {
		netdev_info(priv->ndev,
			    "RX-%d: FIFO overflow.\n", ring->nr);
	}
```

marckleinebudde, 2020-02-29:
> > My driver uses up to 32 RX Objects, that is one complete FIFO. But the error message is misleading. See above. The current driver has that message fixed.
> 
> but the message I see is still the same. Does that mean we're not hitting the RX-MAB overflow?

I've adjusted the text of the RX FIFO and RX MAB overflow messages, so if you still see the exact same text, make sure you've compiled and bootet the correct branch.

jbakuwel, 2020-02-29:
@marckleinebudde Now I'm a bit confused:

$ git branch
  mcp25xxfd-20200211-3
\* mcp25xxfd-20200211-4
  v4.19-rpi/mcp25xxfd-20200211-2
$

$ grep "MAB overflow" drivers/net/can/spi/mcp25xxfd/mcp25xxfd-core.c
	/* RX MAB overflow
$

so it seems I'm on the right branch but the code change you're referring to is not there?

jbakuwel, 2020-02-29:
Note sure what happened with the branches there, so started from scratch and now I can see the changes in the source. Will run the test and report back.

jbakuwel, 2020-02-29:
Hi @marckleinebudde so far not a single "RX-FIFO overflow in FIFO 2" messages in the log. I guess somehow mcp25xxfd-20200211-3 didn't stick either. Test running without a single error message for 30 mins now :-)

[141 more replies]

closed 2024-10-15 by baorepo