Seeed-Studio/pi-hats #15: CAN-HAT can1 can not receive data after some time

jiangfenghua123, 2020-09-24:
Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152284] Internal error: Oops: 5 [#1] PREEMPT SMP ARM

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152550] Process irq/168-mcp25xx (pid: 1035, stack limit = 0xf89bc868)

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152556] Stack: (0xab5e5c60 to 0xab5e6000)

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152568] 5c60: b8ed9d80 7f814c30 7f814ba4 00000040 ab5e5c94 ab5e5c80 8070b82c 8070c31c

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152581] 5c80: b8ed9d80 7f814c30 ab5e5cb4 ab5e5c98 8070b8a4 8070b808 b8ed9d80 b8ed9d80

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152593] 5ca0: 7f814ba4 00000040 ab5e5ccc ab5e5cb8 7f814c30 8070b864 80d07588 b8ed9d80

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152604] 5cc0: ab5e5cf4 ab5e5cd0 80725078 7f814bb0 7f818090 2f7bb189 ab5e5d0c 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152616] 5ce0: 00000000 ba34f008 ab5e5d0c ab5e5cf8 80725110 80725024 00000000 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152628] 5d00: ab5e5d44 ab5e5d10 80725220 807250d8 ffffe000 ba34f104 802115dc ba34f108

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152640] 5d20: 00000001 00000000 00000040 396c1000 0000012c 80c8e000 ab5e5de4 ab5e5d48

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152652] 5d40: 807264d8 8072516c 80853990 80d08494 80d07588 80ad38f0 80d075b0 ffffe000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152664] 5d60: 80d93a94 ba34f000 80d970dc 396c1000 80d05100 ffffca8f 80853b64 ab5e5d7c

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152675] 5d80: ab5e5d7c ab5e5d84 ab5e5d84 ab5e5d8c ab5e5d8c 00000000 8012704c ffffe000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152687] 5da0: 80d941d4 00000008 00208040 ffffe000 ab5e5de4 2f7bb189 80c84348 80c84348

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152699] 5dc0: 0000000c 00000003 80d07d14 80d93c84 00208040 ffffe000 ab5e5e3c ab5e5de8

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152711] 5de0: 801270b0 80726274 8071ca08 80211544 00000025 80a6c414 80aa8da4 80d03080

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152723] 5e00: 80d0308c 80aa8d98 80c8d100 00000000 ab5e5e3c ab5e4000 b8ed9d80 80d07588

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152736] 5e20: b49a8e9c b49a8000 00000000 00000040 ab5e5e54 ab5e5e40 801273e8 80126e2c

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152748] 5e40: 00000000 b8ed9d80 ab5e5e6c ab5e5e58 80723c88 8012735c 0000000f b8ed9d80

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152759] 5e60: ab5e5eac ab5e5e70 7f6dbb28 80723c54 fffffff8 0000000f b5f4b908 2f7bb189

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152771] 5e80: 80853efc b49a8b04 b49a8580 0000000b 7f6e105c b49a8694 00000001 ffff0ff3

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152783] 5ea0: ab5e5f04 ab5e5eb0 7f6db164 7f6db9b8 80512c20 b4ab62e0 b49a8c10 00000002

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152795] 5ec0: 7f6dfcb0 7f6dfc80 80d07588 7f6df438 b4ab6000 2f7bb189 ab5e5f24 acfa2980

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152807] 5ee0: b74c6200 acfa29a4 ffffe000 00000000 80187b64 00000001 ab5e5f24 ab5e5f08

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152819] 5f00: 80187b90 7f6daac4 b74c6200 acfa2980 acfa29a4 ffffe000 ab5e5f74 ab5e5f28

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152830] 5f20: 80187ee8 80187b70 80a71b10 80d07588 80d937a6 8014cd28 ab5e4000 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152842] 5f40: 80187cc0 2f7bb189 80143720 b774d480 acfa2580 00000000 ab5e4000 acfa2980

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152854] 5f60: 80187da0 ab669734 ab5e5fac ab5e5f78 80143df0 80187dac b774d49c b774d49c

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152865] 5f80: 00000000 acfa2580 80143c84 00000000 00000000 00000000 00000000 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152876] 5fa0: 00000000 ab5e5fb0 801010b4 80143c90 00000000 00000000 00000000 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152886] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.152897] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  163.153340] Code: caffffeb e5960008 e3500000 0a000003 (e5904000) 

Message from syslogd@raspberrypi at Sep  7 03:23:28 ...
 kernel:[  164.595616] Disabling IRQ #168

turmary, 2020-09-25:
Driver here are obsolete.
https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/master/modules/CAN-HAT
is the lastest code and always be maintained.

jiangfenghua123, 2020-09-28:
list:
git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays
cd seeed-linux-dtoverlays//modules/CAN-HAT
sudo ./install.sh 
sudo reboot
the branch is master.
resullt:
pi@raspberrypi:~ $ dmesg |grep spi
[    7.729108] mcp25xxfd spi1.0: MCP2517 successfully initialized.
[   10.658199] mcp25xxfd spi1.0 can1: renamed from can0
[   14.553887] Modules linked in: 8021q garp stp llc binfmt_misc brcmfmac brcmutil sha256_generic cfg80211 rfkill raspberrypi_hwmon hwmon snd_bcm2835(C) snd_pcm bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem snd_timer bcm2835_mmal_vchiq(C) v4l2_common videobuf2_dma_contig videobuf2_vmalloc snd videobuf2_memops videobuf2_v4l2 spi_bcm2835aux videobuf2_common videodev vc_sm_cma(C) media spi_bcm2835 uio_pdrv_genirq fixed uio mcp25xxfd can_dev i2c_dev ip_tables x_tables ipv6
pi@raspberrypi:~ $ ifconfig -a
can1: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

jiangfenghua123, 2020-09-28:
├── spi0.0 -> ../../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.0
├── spi0.1 -> ../../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1
└── spi1.0 -> ../../../devices/platform/soc/3f215080.spi/spi_master/spi1/spi1.0

closed 2024-10-15 by baorepo