Seeed-Studio/pi-hats #18: Heavy traffic makes errors and occurs frame drops (low Hz)

ingjae, 2023-02-06:
Hello , @marckleinebudde 

I opened new issue https://github.com/Seeed-Studio/pi-hats/issues/7#issuecomment-1419203361
Thanks for reply 

I'm using rpi4 and 5.4.140-rt64 kernel + 2ch CAN bus FD shield
```
Linux ubuntu 5.4.140-rt64 #1 SMP PREEMPT_RT Sat Oct 16 17:57:52 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

```

and I follow [this](https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/) document 

after heavy traffic , error frames are detected(PCAN-view) , and motor driver detect something wrong (bus off)
when I use **PCAN(Peak CAN)** with socketCAN I can get Hz (control and get feedback)  I want (Under 1Mbps), 
But using **2ch CAN bus FD shield**,  It has frame drops(low Hz) and Eventually, bus off motor with the lowest priority (same code)

here is  my dmesg
```
[    0.000000] Linux version 5.4.140-rt64 (runner@fv-az42-775) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #1 SMP PREEMPT_RT Sat Oct 16 17:57:52 UTC 2021 (Ubuntu 5.4.0-1044.48-raspi 5.4.140)
[    3.225849] spi-bcm2835 fe204000.spi: could not get clk: -517
[   10.937930] spi_master spi0: will run message pump with realtime priority
[   11.022588] mcp251xfd spi0.1 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:0.00MHz) successfully initialized.
[   11.110721] mcp251xfd spi0.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:0.00MHz) successfully initialized.
[  763.834108] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 90 49 f0, CRC=0x5d94) retrying.
[ 1014.408947] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 e1 b1 45, CRC=0xc8f1) retrying.
[ 1145.472032] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=00 24 2b 7e, CRC=0x9b19) retrying.
[ 1162.026867] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 3f a3 a5, CRC=0x282a) retrying.
[ 1162.045392] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=00 8d ae a5, CRC=0x0ffe) retrying.
[ 1162.062180] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=00 cc b8 a5, CRC=0xfee9) retrying.
[ 1162.085613] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 1a c7 a5, CRC=0xf2eb) retrying.
[ 1162.095969] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 6c cd a5, CRC=0x4850) retrying.
[ 1162.096003] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 72 cd a5, CRC=0x49c8) retrying.
[ 1162.108837] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=00 46 d5 a5, CRC=0x1a64) retrying.
[ 1162.183280] mcp251xfd spi0.1 can0: CRC read error at address 0x0010 (length=4, data=80 b6 02 a6, CRC=0xe49e) retrying.

```

here is ```candump any,0~0,#FFFFFFFF -cexdtA```
<details>
<summary> candump </summary>
<div markdown="1">
```
(2023-02-06 16:04:21.785296)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:04:21.785374)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:04:21.785376)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:04:21.785849)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:04:21.785853)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:04:21.785856)  can0  RX - -  20000004   [8]  00 20 00 00 00 00 80 00   ERRORFRAME
	controller-problem{tx-error-passive}
	error-counter-tx-rx{{128}{0}}
 (2023-02-06 16:04:21.785858)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:04:21.785861)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 7E 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{126}{0}}
 (2023-02-06 16:04:21.790969)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5D 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{93}{0}}
[... 68 lines cut ...]
</details>

==================================================================

and I checked phase-seg2 value ``` ip -details link show can0``` , which is 10 so, I set sjw as 5
```
ubuntu@ubuntu:~$ ip -details link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 72 qdisc pfifo_fast state UP mode DEFAULT group default qlen 65536
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <BERR-REPORTING,FD> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 1000 
	  bitrate 1000000 sample-point 0.750 
	  tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  dbitrate 8000000 dsample-point 0.600 
	  dtq 25 dprop-seg 1 dphase-seg1 1 dphase-seg2 2 dsjw 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 40000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 

```

```
sudo ip link set can0 up type can bitrate 1000000 sjw 5   dbitrate 8000000 restart-ms 1000 berr-reporting on fd on && sudo ifconfig can0 txqueuelen 65536
```
and tested again with ```candump any,0~0,#FFFFFFFF -cexdtA```
it returns same result above and **still same problem**

marckleinebudde, 2023-02-06:
Are you sure you want a data bitrate of 8 Mbit/s?

marckleinebudde, 2023-02-06:
What's the output of the `candump` if the error occurs?

ingjae, 2023-02-06:
I want 1Mbit/s CAN2.0 , 

```candump any,0~0,#FFFFFFFF -cexdtA```
```
(2023-02-06 16:15:05.259654)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259668)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259670)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259672)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259676)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259677)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259679)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 78 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{120}{0}}
 (2023-02-06 16:15:05.259681)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.259686)  can0  RX - -  20000004   [8]  00 20 00 00 00 00 80 00   ERRORFRAME
	controller-problem{tx-error-passive}
	error-counter-tx-rx{{128}{0}}
 (2023-02-06 16:15:05.259693)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 7F 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{127}{0}}
 (2023-02-06 16:15:05.265563)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5F 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{95}{0}}
 (2023-02-06 16:15:05.299033)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.299044)  can0  RX - -  20000088   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	protocol-violation{{}{}}
	bus-error
 (2023-02-06 16:15:05.365794)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.365804)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.365806)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.365808)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.365810)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.365812)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 78 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{120}{0}}
 (2023-02-06 16:15:05.366130)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.370745)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5F 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{95}{0}}
 (2023-02-06 16:15:05.495475)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
[... 19 lines cut ...]
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.562512)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 70 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{112}{0}}
 (2023-02-06 16:15:05.562513)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:15:05.562519)  can0  RX - -  20000004   [8]  00 20 00 00 00 00 80 00   ERRORFRAME
	controller-problem{tx-error-passive}
	error-counter-tx-rx{{128}{0}}
 (2023-02-06 16:15:05.562523)  can0  RX - -  20000088   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	protocol-violation{{}{}}
	bus-error
 (2023-02-06 16:15:05.562524)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 7F 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{127}{0}}
 (2023-02-06 16:15:05.563048)  can0  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-06 16:15:05.567193)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5F 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{95}{0}}

```

marckleinebudde, 2023-02-06:
> I want 1Mbit/s CAN2.0

Then remove `dbitrate 8000000` and `fd on`.

ingjae, 2023-02-06:
```sudo ip link set can0 up type can bitrate 1000000 sjw 5 restart-ms 1000 berr-reporting on && sudo ifconfig can0 txqueuelen 65536```

I already tried it before 
anyway I tested again above command line but it has same result 
[image]

```
ubuntu@ubuntu:~$ candump any,0~0,#FFFFFFFF -cexdtA
 (2023-02-06 16:53:12.330152)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:53:12.330164)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 16:53:12.330166)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error

...

```

marckleinebudde, 2023-02-06:
It's still in CAN-FD mode (`mtu 72`).

marckleinebudde, 2023-02-06:
However CAN-FD configuration should be unrelated to CAN TX errors. Have you connected both `can0` and `can1` to the drive?

ingjae, 2023-02-06:
sorry, I had mistake (```mtu 16```) 
[image]
it still have same problem 

Is the drive you are talking about a motor drive? Then I'm using it by connecting the motor drive with only can0 up.
can1 is in no load state

marckleinebudde, 2023-02-06:
ACK - I mean the motor drive.

See if connecting, configuring and bringing up both `can0` and `can1` makes a difference.

ingjae, 2023-02-06:
[image]
I connected both can0 and can1 and it is connected to motor drive seperately
and still have same problem
```
ubuntu@ubuntu:~$ candump any,0~0,#FFFFFFFF -cexdtA
 (2023-02-06 17:43:57.942897)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-06 17:43:57.942911)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
	
...

```

marckleinebudde, 2023-02-06:
Please connect both `can0` and `can1` to the same drive.

ingjae, 2023-02-07:
[image]

The test is carried out in the settings as shown in the figure above.

```
ubuntu@ubuntu:~$ candump any,0~0,#FFFFFFFF -cexdtA
 (2023-02-07 03:34:30.495143)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.496221)  can0  RX - -  20000088   [8]  00 00 94 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error,tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.496243)  can1  RX - -  20000088   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	protocol-violation{{}{}}
	bus-error
 (2023-02-07 03:34:30.497436)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 74 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{116}{0}}
 (2023-02-07 03:34:30.502012)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5C 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{92}{0}}
 (2023-02-07 03:34:30.505216)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.506608)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.506971)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 75 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{117}{0}}
 (2023-02-07 03:34:30.512592)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5C 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{92}{0}}
 (2023-02-07 03:34:30.533190)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.534559)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.539437)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.539914)  can0  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.610938)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.611213)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.672215)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.672565)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.694246)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.695159)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.770218)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.771117)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.815148)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.815881)  can0  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.822820)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.824011)  can0  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.842207)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.842891)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.897905)  can0  RX - -  20000088   [8]  00 00 90 00 00 00 00 00   ERRORFRAME
	protocol-violation{{tx-recessive-bit-error,error-on-tx}{}}
	bus-error
 (2023-02-07 03:34:30.898421)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.898619)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.898923)  can0  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.899248)  can0  RX - -  20000004   [8]  00 08 00 00 00 00 76 00   ERRORFRAME
	controller-problem{tx-error-warning}
	error-counter-tx-rx{{118}{0}}
 (2023-02-07 03:34:30.899539)  can1  RX - -  20000088   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	protocol-violation{{}{}}
	bus-error
 (2023-02-07 03:34:30.900895)  can0  RX - -  20000088   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	protocol-violation{{}{}}
	bus-error
 (2023-02-07 03:34:30.905160)  can0  RX - -  20000004   [8]  00 40 00 00 00 00 5B 00   ERRORFRAME
	controller-problem{back-to-error-active}
	error-counter-tx-rx{{91}{0}}
 (2023-02-07 03:34:30.942135)  can1  RX - -  20000088   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	protocol-violation{{bit-stuffing-error}{}}
	bus-error
 (2023-02-07 03:34:30.942854)  can1  RX - -  20000088   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	protocol-violation{{}{}}
	bus-error

```

marckleinebudde, 2023-02-07:
From the data we have here, I think that one of the motors sends bad CAN messages or you have an electromagnetic interference problem. Can you send me the output of the following commands after the problem occurred:

`ip -s -details link show can0; ip -s -details link show can1``

ingjae, 2023-02-07:
after 10s
```
ubuntu@ubuntu:~$ ip -s -details link show can0; ip -s -details link show can1
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 65536
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <BERR-REPORTING> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 1000 
	  bitrate 1000000 sample-point 0.750 
	  tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 40000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          31         0          13         2          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    144874     31032    3       0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    62031      20667    28      0       0       0       
4: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 65536
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <BERR-REPORTING> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 1000 
	  bitrate 1000000 sample-point 0.750 
	  tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 40000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          32         0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    145187     31120    31      0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    0          0        0       0       0       0       

```

after 30s
```
ubuntu@ubuntu:~$ ip -s -details link show can0; ip -s -details link show can1
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 65536
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <BERR-REPORTING> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 1000 
	  bitrate 1000000 sample-point 0.750 
	  tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 40000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          77         0          29         4          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    339770     72739    9       0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    145250     48406    68      0       0       0       
4: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 65536
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <BERR-REPORTING> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 1000 
	  bitrate 1000000 sample-point 0.750 
	  tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 40000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          81         0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    339920     72817    77      0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    0          0        0       0       0       0       

```

ingjae, 2023-02-07:
[image]

**program(500Hz), target : [SPI-CAN] can0 or can1** -> low Hz , motor drive error 
(At first, the Hz drops, and after a while, there is no response anymore,
Occurs from the motor with the lowest priority (higher id))

**program(500Hz), target : [USB-PCAN] can2** -> each motors works well (500Hz)

**program(100 ~ 200Hz), target : can0, can1 ,can2** -> all works well   (100 ~ 200Hz)

marckleinebudde, 2023-02-07:
Does your program send with the same CAN-ID as one of the motors?

ingjae, 2023-02-07:
Motor 1   -> CAN-ID : 1
Motor 2  -> CAN-ID : 2
Motor 3  -> CAN-ID : 3

program send **feedback request** and **control request**  for  each Motors , ID 1 ,2 ,3
and each Motor drive send **feedback response** (ID 1, 2 ,3)

I checked each Motor's feedback response timestamp , and calculate Hz 

Feedback request is expressed as a cansend, it looks like this: ```motor_id#request_protocol```
```
$ cansend  001#07 
```

so program send **feedback request** like this 
```
001#07
002#07
003#07
```
**control request**  
```
001#05.AA
002#05.BB
003#05.CC
```

receive **feedback response** and parsing, (no control response )
```
001#07.AA.AA.AA.AA.AA.AA.AA
001#07.BB.DD.AA.AA.AA.AA.AA
001#07.CC.AA.AA.AA.AA.AA.AA
```

marckleinebudde, 2023-02-07:
If you send the same CAN-ID from multiple CAN controllers, you can run into collisions on the CAN bus. Which results in the bus errors you see above. Don't know how your drive behaves if it runs into bus errors....

ingjae, 2023-02-07:
No, I send frame at **only one CAN controller** 
I always tested program's target can bus 
**other CANBUS just receive data**

marckleinebudde, 2023-02-07:
Does the motor drive have a CAN controller?
Does the motor drive CAN controller send data?

ingjae, 2023-02-07:
yes, If I send request frame at the SPI CAN controller , then motor drive send response frame

marckleinebudde, 2023-02-07:
So you have 2 CAN controllers sending on the same CAN-ID. You'll run into trouble if you have both controllers send a CAN frame with the same CAN-ID at the same time.

ingjae, 2023-02-08:
So, PCAN accepts what you said and works fine, but SPI CAN causes problems when the response of the motor and the request of the Raspberry Pi overlap for the same ID?

ingjae, 2023-02-08:
Thank you for the answers you have given so far @marckleinebudde  :bow: 
I’m not familiar at CAN structure , sorry for your confusion 
more clearly, In case A in the figure below, it is normal , and stable
are you saying that problems can occur when the time frames of tx and rx overlap, like in case B or case C?
[image]

If so, then the program should send a feedback request, **check the response** to the feedback request, 
and then either issue a control request or send a feedback request?

marckleinebudde, 2023-02-08:
> I’m not familiar at CAN structure , sorry for your confusion more clearly, In case A in the figure below, it is normal, and stable are you saying that problems can occur when the time frames of tx and rx overlap, like in case B or case C?

ACK - if RX and TX use the same CAN-ID.

In your system the higher prio CAN message prevent to low prio messages to be send. After the bus is free, both controllers (`can0` and the drive) send on the same CAN-ID and they collide. 😢

> If so, then the program should send a feedback request, check the response to the feedback request, and then either issue a control request or send a feedback request?

ACK

marckleinebudde, 2023-02-08:
Can you configure the motor to periodically send the information you need?

ingjae, 2023-02-08:
then, I have question ...
How does PCAN(Peak CAN) works on the same system without errors or crashes?
It may be the some technology, but it worked in PCAN and not in SPICAN, so I thought it was a problem with SPI CAN.

marckleinebudde, 2023-02-08:
I don't know what the PCAN adapter does, it's a µC with firmware. I don't know if all events are forwarded via USB and if the Linux driver understands all and forwards them to user space.

Besides from that, the timing with the USB adapter might be different compared to the SPI adapter.

ingjae, 2023-02-08:
> Can you configure the motor to periodically send the information you need?

Current motors don't have that feature and I have to ask for it
[image]

ingjae, 2023-02-08:
If the motor is configured to periodically give the desired information, how do I avoid overlapping when I send control frames?

marckleinebudde, 2023-02-08:
I think you cannot. Does that happen as often as the periodically messages?

ingjae, 2023-02-08:
did you mean can't avoid overlapping when I send control frames?
I have other motors that periodically send feedback messages.

my program's above cycle target is 500Hz 
but, in ```candump can0 -ta ``` actual Hz is low (about 250Hz)

marckleinebudde, 2023-02-08:
The design problem is that there are 2 senders with the same CAN-ID.

If the motor sends periodically you cannot avoid overlaps if you send from `can0`.

If you motor only sends after you've requested to send, then:
1. request information from motor
2. wait for answer from motor
3. send something else to motor (e.g.: control message)

marckleinebudde, 2023-02-08:
> If I understand it right, does it mean that frames with different IDs can be received from one motor at the same time?

If multiple senders try to send different CAN-ID at the same time, the message with the lowest CAN-ID will be send on the bus. The controllers with the other messages will retry later, there will be _no_ errors.

> In order to send and receive both control and feedback protocols at the same time, the protocol itself needs more information in the id like canopen, right?

That's an option, but you only have to avoid the same CAN-ID from multiple senders.

ingjae, 2023-02-08:
Thank you so much @marckleinebudde 
 I understood why can ID should not overlap while looking at the priority check part.
[image]
If multiple senders send the same ID at the same time, each controller determines that they can send each other, and a collision occurs.

SPI CAN is not a problem, it is my system problem Specifically, current protocol.

marckleinebudde, 2023-02-08:
In this picture node 16 will stop transmitting, and retry later - without a error. If the collision occurs _after_ the CAN-ID it's a _real_ collision with errors.

closed 2023-05-09 by ingjae